WeekendWarrior1/XTronical_XT_DAC_Audio_Mirror #1: Error compiling for ESP32

ELNAHUAL, 2021-04-20:
Every time I try to use tis library I got the "Error compiling for board ESP32 Dev Module" it doesn't matter if I choose different ESP32 board, I'm having same error for each ESP32 board.
If I choose an older library it does compile and load to the board but there's no sound at all at the DAC pin.
Maybe something changed on the espressif repository from Github.
Can anyone help?
I'll appreciate it your kind help.

beer73, 2021-11-12:
Yeah, same problem here, please, someone to light the way.
Or tell us which version of esp32 and library has worked for you.
I use an ESP WROOM 32 DEVKITV1.

beer73, 2021-11-12:
Ups, I downgraded the ESP32 library from 2.0.1 to 1.0.6 and all fine.
Sometimes these little things seem like mountains to newbies like me.

rppsys, 2021-12-16:
This is the error:

    Compiling libraries...
    Compiling library "XT_DAC_Audio"
    
    (...)
    
    /home/arduino/libraries/XT_DAC_Audio/XT_DAC_Audio.cpp: In member function 'void XT_Instrument_Class::SetWaveForm(uint8_t)':
    /home/arduino/libraries/XT_DAC_Audio/XT_DAC_Audio.cpp:667:18: error: deleting object of polymorphic class type 'XT_Wave_Class' which has non-virtual destructor might cause undefined behaviour [-Werror=delete-non-virtual-dtor]
      delete (WaveForm);          // free any previous memory for a previous waveform
                      ^
    cc1plus: some warnings being treated as errors
    Usando biblioteca XT_DAC_Audio na pasta: /home/arduino/libraries/XT_DAC_Audio (legacy)
    exit status 1
    Erro compilando para a placa DOIT ESP32 DEVKIT V1

syndicated-8, 2022-03-19:
I'm getting this error after installing the library for ESP32 WROOM 32S

In file included from C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/esp32s2/include/hal/include/hal/touch_sensor_types.h:11,
                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/esp32s2/include/esp_hw_support/include/esp_sleep.h:12,
                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal.h:33,
                 from C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-timer.h:23,
                 from C:\Users\DevEnv\Documents\Arduino\libraries\XT_DAC_Audio\Mixing\XT_DAC_Audio.cpp:23:
C:\Users\DevEnv\Documents\Arduino\libraries\XT_DAC_Audio\Mixing\XT_DAC_Audio.cpp: In function 'void onTimer()':
XT_DAC_Audio.cpp:155:31: error: 'RTC_IO_PAD_DAC1_REG' was not declared in this scope

RobKeij, 2022-04-16:
Also here but I got more errors, to many to list here.
I'm trying to find the missing declarations etc.

k1dbl4ck, 2022-06-10:
@syndicated-8 / @RobKeij 

`RTC_IO_PAD_DAC1_REG` is declared in the esp32 soc headers....

Solve by adding `#include "soc/rtc_io_reg.h"` to the top of `XT_DAC_Audio.cpp`

WeekendWarrior1, 2022-06-12:
> @syndicated-8 / @RobKeij
> 
> `RTC_IO_PAD_DAC1_REG` (and other io registers) is declared in the esp32 soc headers....
> 
> Solve by adding `#include "soc/rtc_io_reg.h"` to the top of `XT_DAC_Audio.cpp`

Thanks @k1dbl4ck, I have committed your fix.

closed 2022-06-12 by WeekendWarrior1