WeekendWarrior1/XTronical_XT_DAC_Audio_Mirror #6: Any clue on this error? I using the latest version(4.2.1) but had this issue. My board is ESP32 and platform using is arduino.

hongwansoon0821, 2023-04-12:
c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:26:10: error: #include expects "FILENAME" or <FILENAME>
 #include "soc/rtc_io_reg.h
          ^~~~~~~~~~~~~~~~~
In file included from C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:19,
                 from C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/touch_sensor_types.h:11,
                 from C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/esp_hw_support/include/esp_sleep.h:12,
                 from C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/esp32-hal.h:33,
                 from C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/esp32-hal-timer.h:23,
                 from c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:23:
c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp: In function 'void onTimer()':
c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:156:31: error: 'RTC_IO_PAD_DAC1_REG' was not declared in this scope
             SET_PERI_REG_BITS(RTC_IO_PAD_DAC1_REG, RTC_IO_PDAC1_DAC, LastDacValue, RTC_IO_PDAC1_DAC_S);
                               ^~~~~~~~~~~~~~~~~~~
C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/esp_common/include/esp_assert.h:29:38: note: in definition of macro 'TRY_STATIC_ASSERT'
             if (__builtin_constant_p(CONDITION) && !(CONDITION)) {          \
                                      ^~~~~~~~~
C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:96:57: note: in expansion of macro 'IS_DPORT_REG'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                                         ^~~~~~~~~~~~
C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:205:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((reg), SET_PERI_REG_BITS);                                                             \
             ^~~~~~~~~~~~~~~~~~~
c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:156:13: note: in expansion of macro 'SET_PERI_REG_BITS'
             SET_PERI_REG_BITS(RTC_IO_PAD_DAC1_REG, RTC_IO_PDAC1_DAC, LastDacValue, RTC_IO_PDAC1_DAC_S);
             ^~~~~~~~~~~~~~~~~
In file included from c:\users\itadm\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\sys-include\sys\reent.h:503,
                 from C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/newlib/platform_include/sys/reent.h:17,
                 from c:\users\itadm\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch5\xtensa-esp32-elf\sys-include\stdio.h:60,
                 from C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/esp32-hal.h:25,
                 from C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/esp32-hal-timer.h:23,
                 from c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:23:
c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:156:31: error: 'RTC_IO_PAD_DAC1_REG' was not declared in this scope
             SET_PERI_REG_BITS(RTC_IO_PAD_DAC1_REG, RTC_IO_PDAC1_DAC, LastDacValue, RTC_IO_PDAC1_DAC_S);
                               ^~~~~~~~~~~~~~~~~~~
C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:96:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^~~~~~~~~~~~~~~~~
C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:96:57: note: in expansion of macro 'IS_DPORT_REG'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                                         ^~~~~~~~~~~~
C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:205:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((reg), SET_PERI_REG_BITS);                                                             \
             ^~~~~~~~~~~~~~~~~~~
c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:156:13: note: in expansion of macro 'SET_PERI_REG_BITS'
             SET_PERI_REG_BITS(RTC_IO_PAD_DAC1_REG, RTC_IO_PDAC1_DAC, LastDacValue, RTC_IO_PDAC1_DAC_S);
             ^~~~~~~~~~~~~~~~~
In file included from C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:19,
                 from C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/hal/include/hal/touch_sensor_types.h:11,
                 from C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/esp_hw_support/include/esp_sleep.h:12,
                 from C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/esp32-hal.h:33,
                 from C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/esp32-hal-timer.h:23,
                 from c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:23:
c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:156:31: error: 'RTC_IO_PAD_DAC1_REG' was not declared in this scope
             SET_PERI_REG_BITS(RTC_IO_PAD_DAC1_REG, RTC_IO_PDAC1_DAC, LastDacValue, RTC_IO_PDAC1_DAC_S);
                               ^~~~~~~~~~~~~~~~~~~
C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/esp_common/include/esp_assert.h:29:38: note: in definition of macro 'TRY_STATIC_ASSERT'
             if (__builtin_constant_p(CONDITION) && !(CONDITION)) {          \
                                      ^~~~~~~~~
C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:96:57: note: in expansion of macro 'IS_DPORT_REG'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                                         ^~~~~~~~~~~~
C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:175:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((addr), WRITE_PERI_REG);                                                               \
             ^~~~~~~~~~~~~~~~~~~
C:\Users\itadm\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/soc/esp32/include/soc/soc.h:206:14: note: in expansion of macro 'WRITE_PERI_REG'
             (WRITE_PERI_REG((reg),(READ_PERI_REG(reg)&(~((bit_map)<<(shift))))|(((value) & bit_map)<<(shift)) ));      \
              ^~~~~~~~~~~~~~
c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:156:13: note: in expansion of macro 'SET_PERI_REG_BITS'
             SET_PERI_REG_BITS(RTC_IO_PAD_DAC1_REG, RTC_IO_PDAC1_DAC, LastDacValue, RTC_IO_PDAC1_DAC_S);
             ^~~~~~~~~~~~~~~~~
[... 440 lines cut ...]
c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:163:13: note: in expansion of macro 'SET_PERI_REG_MASK'
             SET_PERI_REG_MASK(RTC_IO_PAD_DAC2_REG, RTC_IO_PDAC2_XPD_DAC | RTC_IO_PDAC2_DAC_XPD_FORCE);
             ^~~~~~~~~~~~~~~~~
c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:163:52: error: 'RTC_IO_PDAC2_XPD_DAC' was not declared in this scope
[... cut ...]
             (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val);                                       \
                                                                            ^~~
c:\Users\itadm\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:163:13: note: in expansion of macro 'SET_PERI_REG_MASK'
             SET_PERI_REG_MASK(RTC_IO_PAD_DAC2_REG, RTC_IO_PDAC2_XPD_DAC | RTC_IO_PDAC2_DAC_XPD_FORCE);
             ^~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

wilson3682, 2023-10-29:
@hongwansoon0821 I added the following piece of code to XT_DAC_Audio.cpp and It is working now for me.

#include "soc/rtc_io_reg.h"

Reference link:
https://www.esp32.com/viewtopic.php?t=27178#p99673

joex92, 2025-06-25:
hi, after adding that line, I get this error:

```console
loading library from d:\Documents\Arduino\libraries\Joystick: invalid library: no header files foundloading library from d:\Documents\Arduino\libraries\Joystick2: invalid library: no header files foundloading library from d:\Documents\Arduino\libraries\Joystick3: invalid library: no header files foundloading library from d:\Documents\Arduino\libraries\lufa-LUFA-210130: invalid library: no header files found
FQBN: esp32:esp32:esp32da
Using board 'esp32da' from platform in folder: C:\Users\jandy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0
Using core 'esp32' from platform in folder: C:\Users\jandy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.2.0

cmd /c if exist "C:\\Users\\JxM92\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2025525-6644-1n8bfff.s9o1\\WavSpeed\\partitions.csv" COPY /y "C:\\Users\\JxM92\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2025525-6644-1n8bfff.s9o1\\WavSpeed\\partitions.csv" "C:\\Users\\JxM92\\AppData\\Local\\arduino\\sketches\\4F02CF1F2DE25EBA23CCA111D9ACBC98\\partitions.csv"
cmd /c if not exist "C:\\Users\\JxM92\\AppData\\Local\\arduino\\sketches\\4F02CF1F2DE25EBA23CCA111D9ACBC98\\partitions.csv" if exist "C:\\Users\\jandy\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\variants\\esp32da\\partitions.csv" COPY "C:\\Users\\jandy\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\variants\\esp32da\\partitions.csv" "C:\\Users\\JxM92\\AppData\\Local\\arduino\\sketches\\4F02CF1F2DE25EBA23CCA111D9ACBC98\\partitions.csv"
cmd /c if not exist "C:\\Users\\JxM92\\AppData\\Local\\arduino\\sketches\\4F02CF1F2DE25EBA23CCA111D9ACBC98\\partitions.csv" COPY "C:\\Users\\jandy\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\tools\\partitions\\default.csv" "C:\\Users\\JxM92\\AppData\\Local\\arduino\\sketches\\4F02CF1F2DE25EBA23CCA111D9ACBC98\\partitions.csv"
cmd /c IF EXIST "C:\\Users\\JxM92\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2025525-6644-1n8bfff.s9o1\\WavSpeed\\bootloader.bin" ( COPY /y "C:\\Users\\JxM92\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2025525-6644-1n8bfff.s9o1\\WavSpeed\\bootloader.bin" "C:\\Users\\JxM92\\AppData\\Local\\arduino\\sketches\\4F02CF1F2DE25EBA23CCA111D9ACBC98\\WavSpeed.ino.bootloader.bin" ) ELSE ( IF EXIST "C:\\Users\\jandy\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\variants\\esp32da\\bootloader.bin" ( COPY "C:\\Users\\jandy\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.2.0\\variants\\esp32da\\bootloader.bin" "C:\\Users\\JxM92\\AppData\\Local\\arduino\\sketches\\4F02CF1F2DE25EBA23CCA111D9ACBC98\\WavSpeed.ino.bootloader.bin" ) ELSE ( "C:\\Users\\jandy\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esptool_py\\4.9.dev3\\esptool.exe" --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "C:\\Users\\JxM92\\AppData\\Local\\arduino\\sketches\\4F02CF1F2DE25EBA23CCA111D9ACBC98\\WavSpeed.ino.bootloader.bin" "C:\\Users\\jandy\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.4-2f7dcd86-v1\\esp32\\bin\\bootloader_qio_80m.elf" ) )
esptool.py v4.8.1
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
cmd /c if exist "C:\\Users\\JxM92\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2025525-6644-1n8bfff.s9o1\\WavSpeed\\build_opt.h" COPY /y "C:\\Users\\JxM92\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2025525-6644-1n8bfff.s9o1\\WavSpeed\\build_opt.h" "C:\\Users\\JxM92\\AppData\\Local\\arduino\\sketches\\4F02CF1F2DE25EBA23CCA111D9ACBC98\\build_opt.h"
cmd /c if not exist "C:\\Users\\JxM92\\AppData\\Local\\arduino\\sketches\\4F02CF1F2DE25EBA23CCA111D9ACBC98\\build_opt.h" type nul > "C:\\Users\\JxM92\\AppData\\Local\\arduino\\sketches\\4F02CF1F2DE25EBA23CCA111D9ACBC98\\build_opt.h"
cmd /c type nul > "C:\\Users\\JxM92\\AppData\\Local\\arduino\\sketches\\4F02CF1F2DE25EBA23CCA111D9ACBC98/file_opts"
cmd /c COPY /y "C:\\Users\\jandy\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.4-2f7dcd86-v1\\esp32\\sdkconfig" "C:\\Users\\JxM92\\AppData\\Local\\arduino\\sketches\\4F02CF1F2DE25EBA23CCA111D9ACBC98\\sdkconfig"
        1 file(s) copied.
Detecting libraries used...
C:\Users\jandy\AppData\Local\Arduino15\packages\esp32\tools\esp-x32\2411/bin/xtensa-esp32-elf-g++ -c @C:\Users\jandy\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32/flags/cpp_flags -w -Os -Werror=return-type -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_WROOM_DA -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_WROOM_DA" -DARDUINO_VARIANT="esp32da" -DARDUINO_PARTITION_default -DARDUINO_HOST_OS="windows" -DARDUINO_FQBN="esp32:esp32:esp32da:UploadSpeed=921600,CPUFreq=240,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=none,LoopCore=1,EventsCore=1,EraseFlash=none" -DESP32=ESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 @C:\Users\jandy\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32/flags/defines -IC:\Users\JxM92\AppData\Local\Temp\.arduinoIDE-unsaved2025525-6644-1n8bfff.s9o1\WavSpeed -iprefix C:\Users\jandy\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32/include/ @C:\Users\jandy\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.4-2f7dcd86-v1\esp32/fla [... cut ...]
d:\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:668:9: warning: deleting object of polymorphic class type 'XT_Wave_Class' which has non-virtual destructor might cause undefined behavior [-Wdelete-non-virtual-dtor]
  668 |         delete (WaveForm);                                                                              // free any previous memory for a previous waveform
      |         ^~~~~~~~~~~~~~~~~
Using library XT_DAC_Audio in folder: d:\Documents\Arduino\libraries\XT_DAC_Audio (legacy)
exit status 1

Compilation error: exit status 1
```

Using a ESP32-WROOM-32

petrbrouzda, 2025-06-25:
> hi, after adding that line, I get this error:
...

Based on your error, you have ESP32 arduino core 3.x - can be checked in "board manager" in  Arduino IDE.
It will not work for sure, there are many differences between ESP32 core 2.x and 3.x.
So change version of your ESP32 board in board manager to latest 2.x (2.0.17) and try again.
(Yes, board version can be changed up and down as you want.)

Library source must be updated to work with ESP32 core 3.x  - and there are no changes for some years.

joex92, 2025-06-25:
oh ok, thanks... I'll use core 2.x...

I suppose updating this to the latest core would be too much work...

petrbrouzda, 2025-06-26:
I'm not the owner...
But now I found somebody just did it:
https://github.com/WeekendWarrior1/XTronical_XT_DAC_Audio_Mirror/pull/13