WeekendWarrior1/XTronical_XT_DAC_Audio_Mirror #9: Compilation error with ESP32

walking, 2024-05-29:
Hi! I'm getting this error when I try to compile. I think it's related to some updates in different libraries...

I'm using ESP32 DEVKITV1

```cpp
/Users/user/Documents/Arduino/libraries/XT_DAC_Audio/XT_DAC_Audio.cpp: In constructor 'XT_DAC_Audio_Class::XT_DAC_Audio_Class(uint8_t, uint8_t, uint16_t)':
/Users/user/Documents/Arduino/libraries/XT_DAC_Audio/XT_DAC_Audio.cpp:305:27: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  305 |         timer = timerBegin(TimerNo, 80, true);           // use timer TimerNo, pre-scaler is 80 (divide by 8000), count up
      |                 ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
In file included from /Users/user/Documents/Arduino/libraries/XT_DAC_Audio/XT_DAC_Audio.cpp:24:
/Users/user/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
/Users/user/Documents/Arduino/libraries/XT_DAC_Audio/XT_DAC_Audio.cpp:306:29: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  306 |         timerAttachInterrupt(timer, &onTimer, true); // P3= edge triggered
      |         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
/Users/user/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
/Users/user/Documents/Arduino/libraries/XT_DAC_Audio/XT_DAC_Audio.cpp:307:9: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  307 |         timerAlarmWrite(timer, 20, true);                        // will trigger 250,000 times per second,
      |         ^~~~~~~~~~~~~~~
      |         timerWrite
/Users/user/Documents/Arduino/libraries/XT_DAC_Audio/XT_DAC_Audio.cpp:308:9: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  308 |         timerAlarmEnable(timer);                                         // enable
      |         ^~~~~~~~~~~~~~~~
      |         timerAlarm

exit status 1

Compilation error: exit status 1
```

walking, 2024-05-31:
the issue happens with esp32 v3.0.0.
Works good with v2.0.17

patrick-kiwi, 2024-07-02:
I'm also having issues trying to get the wavSpeed example to compile with the "esp32 Dev Module" option.  In the board manager I tried esp32 v2.0.17, but that made no difference.  I get a bunch of errors, too many to paste:  

In file included from C:\Users\Patrick\Documents\ArduinoData\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/include/esp_common/include/esp_macros.h:13,
                 from C:\Users\Patrick\Documents\ArduinoData\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/include/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:76,
                 from C:\Users\Patrick\Documents\ArduinoData\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/include/freertos/FreeRTOS-Kernel/include/freertos/portable.h:59,
                 from C:\Users\Patrick\Documents\ArduinoData\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-bd2b9390ef\esp32/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:71,
                 from C:\Users\Patrick\Documents\ArduinoData\packages\esp32\hardware\esp32\3.0.2\cores\esp32/esp32-hal.h:34,
                 from C:\Users\Patrick\Documents\ArduinoData\packages\esp32\hardware\esp32\3.0.2\cores\esp32/esp32-hal-timer.h:25,