WeekendWarrior1/XTronical_XT_DAC_Audio_Mirror #11: Error while compiling

saeedalikhani98, 2024-10-09:
C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp: In constructor 'XT_DAC_Audio_Class::XT_DAC_Audio_Class(uint8_t, uint8_t, uint16_t)':
C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:278:27: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  278 |         timer = timerBegin(TimerNo, 80, true);          // use timer TimerNo, pre-scaler is 80 (divide by 8000), count up
      |                 ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
In file included from C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:10:
C:\Users\saeed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:279:29: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  279 |         timerAttachInterrupt(timer, &onTimer, true);    // P3= edge triggered
      |         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\saeed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:280:9: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  280 |         timerAlarmWrite(timer, 20, true);               // will trigger 250,000 times per second,
      |         ^~~~~~~~~~~~~~~
      |         timerWrite
C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:281:9: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  281 |         timerAlarmEnable(timer);                        // enable
      |         ^~~~~~~~~~~~~~~~
      |         timerAlarm

saeedalikhani98, 2024-10-09:
My problem was not solved by adding this line of code to XT_DAC_Audio.c

#include "soc/rtc_io_reg.h"

saeedalikhani98, 2024-10-09:
I added the library "soc/rtc_io_reg.h". The error changed to

C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp: In member function 'void XT_DAC_Audio_Class::FillBuffer()':
C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:208:27: error: 'DacVolume' was not declared in this scope; did you mean 'SetVolume'?
  208 |             AudioVolume = DacVolume;                                // Audio Volume patch by TEB, Sep-16-2019. 0=Off, 100=Full Volume.
      |                           ^~~~~~~~~
      |                           SetVolume
C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp: In constructor 'XT_DAC_Audio_Class::XT_DAC_Audio_Class(uint8_t, uint8_t, uint16_t)':
C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:288:27: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
  288 |         timer = timerBegin(TimerNo, 80, true);          // use timer TimerNo, pre-scaler is 80 (divide by 8000), count up
      |                 ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
In file included from C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:3:
C:\Users\saeed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
   35 | hw_timer_t *timerBegin(uint32_t frequency);
      |             ^~~~~~~~~~
C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:289:29: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
  289 |         timerAttachInterrupt(timer, &onTimer, true);    // P3= edge triggered
      |         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
C:\Users\saeed\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.4\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
   50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
      |      ^~~~~~~~~~~~~~~~~~~~
C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:290:9: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
  290 |         timerAlarmWrite(timer, 20, true);               // will trigger 250,000 times per second,
      |         ^~~~~~~~~~~~~~~
      |         timerWrite
C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:291:9: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
  291 |         timerAlarmEnable(timer);                        // enable
      |         ^~~~~~~~~~~~~~~~
      |         timerAlarm
C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp: In member function 'virtual uint8_t XT_Sequence_Class::NextByte()':
C:\Users\saeed\Documents\Arduino\libraries\XT_DAC_Audio\XT_DAC_Audio.cpp:1267:12: error: 'ClearAfterPlay' was not declared in this scope
 1267 |         if(ClearAfterPlay && !RepeatForever)// Clear the Sequence list, Start Anew. Steve, Oct-21-2019.
      |            ^~~~~~~~~~~~~~

saeedalikhani98, 2024-10-09:
I added the library "soc/rtc_io_reg.h". The error could not be resolved

TanimSk, 2025-02-20:
because you're using newer version of esp platform, you can decrease the version of your platform to `espressif32@3.5.0` or modify the library with updated APIs.
Here is the updated API doc: https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html

You've to add/modify the following lines in `XT_DAC_Audio.cpp` file:

add these header files
```cpp
#include "driver/dac.h"   // ESP32 DAC driver
#include "soc/rtc_io_reg.h"
```

replace:
```cpp
// Set up interrupt routine
timer = timerBegin(TimerNo, 80, true);          // use timer TimerNo, pre-scaler is 80 (divide by 8000), count up
timerAttachInterrupt(timer, &onTimer, true);    // P3= edge triggered
timerAlarmWrite(timer, 20, true);               // will trigger 250,000 times per second,
timerAlarmEnable(timer);                        // enable
delay(1);                             			// Allow system to settle, otherwise garbage can play for first second
```

with:

```cpp
timer = timerBegin(1000000);		   // Set timer frequency to 1 MHz (1 µs per tick)
timerAttachInterrupt(timer, &onTimer); // Attach interrupt handler
timerAlarm(timer, 20, true, 0);		   // Set up the alarm, enable it automatically, no reload count
delay(1);							   // Allow system to settle
```